iancanada/DocumentDownload #1: Use Battery PSU when battery charger is on?

mbrennwa, 2020-06-30:
Is it possible to keep the battery PSU output active while the batteries are being charged?
If this is possible, how do I configure the PSU for this?

iancanada, 2020-06-30:
Hi ,

This is a pure battery PSU, it will degrade the battery's ultra low noise
performance if being charged while working.
So I don't think it is a good idea. You can not do it that way.

Regards,
Ian

brennmat, 2020-06-30:
I do understand that the best performance is only achieved while the charger is disconnected from the batteries. However, there are applications where continuous power supply is useful (if not required). The charger on / off behaviour could be controlled via an input pin to the battery PSU, for example based on the playback status of the music system.

Example: continuous power supply to a FiFo reclocker module for a Raspberry Pi