iancanada/DocumentDownload #4: TRANSPORTPI DIGI Ultra Low Jitter Digital Audio Interface Module Driver problem

Dittie, 2023-01-17:
Hello 
I am using a rpi with a TRANSPORTPI DIGI Ultra Low Jitter Digital Audio Interface Module on top and a purepi with an ifi ipower adapter. The transportpi sends its signal via toslink in to a beresford caiman seg dac. I placed the accusilicon clocks. The os is volumio. When I am using this setup for a while, say four albums, then it starts to click and pop like an old vinyl record during playing. When I restart it still occures. When I completely shut down then it will play without any problems. After a couple of albums it starts all over again. 

I contacted volumio they think the problem is the driver. I am using the recommended driver: hifiberry digi+ pro. I also tried the hifiberry ADC, but the same problem occurs. The driver for audiophonics sabre and the generic i2s doesn’t work.

Which driver should I use to avoid the problem?

Greetings Dittie

Dittie, 2023-01-17:
ACCUSILICON AS318-B-451584 Ultra Low Jitter Clock 45MHz

I have these clocks, recommended on the audiophonics website. I didn’t used the original clocks but I can try.

Dittie, 2023-01-17:
I already use wired, wifi is off.
I have two clocks of 45mhz, maybe this is the problem? Or is this is an misunderstandig?

iancanada, 2023-01-17:
You have to use 45 and 49. Two with the same frequency will result in a
problem.

Dittie, 2023-01-17:
Then that’s the problem. I have two 45 mhz. So I wil order one 49 mhz. Is that correct?

iancanada, 2023-01-17:
Yes, you figured it out. Please go ahead with ordering the 49.xxx
For now, you can use one 45.xxxMHz accusilicon and one 49.xxxx orignial XO.
Please don't plug them in the wrong socket.

Dittie, 2023-01-17:
Great! Thanks for your support!

How can I see which one is the 49.xxxx original XO?

iancanada, 2023-01-17:
Using a magnifier