iancanada/DocumentDownload #5: Static on I2S output to DAC.

montereytom, 2023-03-22:
Static only on I2S Ouptput on TRANSPORTPI DIGI Ultra Low Jitter Digital Audio Interface Module --- Coaxial and Toslink OK

(1) RPi 4B - 4GB (2019)
(2) Audio-gd R1 Dac (PS Audio Compatible)
(3) AudioQuest Pearl 48 HDMI cable (directional) (also tried generic HDMI 4K cable)
(4) Volumio Version 3.435
(5) Tried HiFiBerry Digi+ Pro and Generic I2S with no success.

What do I need to do to get the I2S output to work with my DAC, e.g., different or custom driver, jumper change on board, etc.? Any assistance you can provide will be greatly appreciated. 

[image]

iancanada, 2023-03-22:
You have to use the HiFiBerry Digi+ Pro  as the driver.
You need to config your DAC to the HDMI input

Please refer to this document for the configuration
https://github.com/iancanada/DocumentDownload/blob/master/HDMIpi/HdmiPiPro/HdmiPiProManual.pdf

Make sure spdif outputs are all good first and then connect the HDMI I2S.

montereytom, 2023-03-22:
Thank you for your quick reply.   
I have done all that you have suggested as well as followed the instruction for this specific card (https://www.audiophonics.fr/en/dac-and-interface-modules/ian-canada-transportpi-digi-ultra-low-jitter-digital-audio-interface-module-p-16714.html#idTab9) but to no avail.  From the attached picture, it looks like one of the clocks is active when the unit is connected via HDMI to my DAC.
Any other suggestions will be greatly appreciated.

iancanada, 2023-03-23:
Can you make sure that the spdif outputs are good as before? Otherwise, it
will be a software issue.
I suggest using Volumio for a start up.

montereytom, 2023-03-23:
Thank you again for replying.  The SPDIF outputs are fine and I am running the latest version of Volumio.  Here is my setup:
(1) RPi 4B - 4GB (2019)
(2) Audio-gd R1 Dac (PS Audio Compatible)
(3) AudioQuest Pearl 48 HDMI cable (directional) (also tried generic HDMI 4K cable)
(4) Volumio Version 3.435 with I2S running HiFiBerry Digi+ Pro.

iancanada, 2023-03-23:
Hi,

If that's the case, I suspect it's the DAC issue. I would suggest trying
other DAC to make sure.

montereytom, 2023-03-24:
Thanks, I'll have to find one with an I2S input option; it may take a while.  Optionally, I may try another I2S hat which may be a more feasible option.  Regardless, I will get back to you with the results.

montereytom, 2023-04-06:
FOLLOW-UP:  I purchased the Douk Audio U2 PRO XMOS XU208 Digital Interface which provides USB to I2S to test the I2S operation of my Audio-gd R1 DAC (and it works).  It has the same pin-out configuration as the  TRANSPORTPI DIGI board with the exception of pin 16 (MUTEn - see attached).  Therefore, since I bought the board (from Audiophonics) specifically for I2S, I would like to return it for a credit or refund.  Do I return it to you or to Audiophonics?  BTW, the coaxial output from the TRANSPORTPI DIGI  board is excellent!
[image]

iancanada, 2023-04-06:
Hi,

Thank you for the update.
Please discuss with Audiophonics if you want to return.