iancanada/DocumentDownload #9: Roon Does Not See TransportPi Digi with RoPieee (Volumio OK)

montereytom, 2023-07-24:
I am running 2 Pi4Bs:
     #1 with Volumio v5 and a Douk Audio USB>I2S interface, and
     #2 with RoPieee (2023.5.1) and an Ian TransportPi Digi
#1 works fine with Volumio with Roon enabled and DAC (Audio-GD - PS Audio compatible) is selectable in “Audio” ALSA choices.
#2 Pi4B with USB output directly into DAC (no hat) will show up as a Roon ALSA choice in Roon “Audio” and works fine, but Digi TransportPi Digi does not show up in “Audio” choices when connected to DAC (I2S or coaxial). TransportPi Digi does work with Volumio v5 (coaxial).
I have tried all recent versions of RoPieee with no success. Any help will be greatly appreciated.

iancanada, 2023-07-25:
For TransportPiDigi, please always select *Hifiberry DIGI+ Pro *as the
linux driver.
Hardware doesn't have any issue as long as you have no problem with Volumio.

montereytom, 2023-07-25:
Thanks for your reply.---Tom Hicks ---