iancanada/DocumentDownload #10: TRANSPORTPI AES running on Volumio+Tidal problem

Joonyahny, 2024-01-05:
Hi 
I have TRANSPORTPI AES running on Volumio+Tidal and having problems with 24bit 192 music streaming. Music is stuttering. I mean it is like stop and go and stop and go situation.  It happens only on 24bit 192 music.  Clock signal is provided by FifoPiQ7.  I have accusilicon clock chips and they showed problems so I changed back to the original parts and the problem persists.  can you help me on this?

iancanada, 2024-01-05:
Can you send me a picture of your system?

Joonyahny, 2024-01-06:
[image]
[image]
[image]
[image]

closed 2024-01-06 by Joonyahny

iancanada, 2024-01-06:
> 20240106_150046.jpg (view on web)
> 20240106_150022.jpg (view on web)
> 20240106_150010.jpg (view on web)
> [image: Uploading 20240106_150001.jpg…]

What's your DAC? Did you have this issue before?

Joonyahny, 2024-01-06:
I am using Brooklyn DAC+. Never had this kind of problem before.

Joonyahny, 2024-01-06:
TransportPI and DAC is connected with 75ohm digital COAX cable.

iancanada, 2024-01-06:
Do you mean the system has been working and has a problem suddenly?
Can you re-connect the MCLK cable? Can you try other DAC to exclude the DAC
issue?

Thanks
Ian

Joonyahny, 2024-01-07:
No. The system is working fine with TransportPI DIGI (I mean any frequency) but it is not working with AES.  I have two units, DIGI and AES+Q7.  AES shows interruptions (stuttering) during streaming of 24bit 192 khz.  Lower than this, it is working fine.  I assume the limit for my AES unit is 24bit 192 khz.  Tried re-connection but problem remains and unfortunately, Brooklyn DAC+ is the only DAC I have.

iancanada, 2024-01-07:
I'm trying to duplicate your case with the same configuration today. But I
don't have any problem playing 192KHz 24bit music with three of my DACs.
All OPT.RCA and AES cables are good.
I really want to address your issue. If you want, you can send both Q7 and
AES back to me to check.

BTW, Q7+AES needs the Audiophoincs ESS DAC as the Linux driver. I think you
have already chosen it correctly.

Please let me know.

Regards
Ian

Joonyahny, 2024-01-07:
Thanks Ian.  Yes I have chosen the Audiophonics driver.  I will send you Q7+AES for check.  Can you send me the address?

iancanada, 2024-01-07:
Please list it as "hobby PCB to repair $10-" on the shipping paperwork to
avoid any additional cost.Please don't forget to put a printed note about
your case and your email, as well as your address.Please let me know if you
have a tracking number.
IanCanada
71 Meadowbank Rd,Etobicoke,
ON M9B 5C7,
Canada
6472273388

Joonyahny, 2024-01-07:
Thanks Ian. Will get back to you soon.

Joonyahny, 2024-01-10:
Hi Ian, just one question.  Can AES play 32 bit 44.1khz?  I just directly connected my dac to the computer and some of the songs that could not played with AES were not 24 bit 192, those were 32 bit 44.1khz.  So maybe I was streaming higher spec file than AES can handle?

iancanada, 2024-01-10:
Try to play true 192KHz/24bit to see if it is OK. If you have a
MonitorPi/Pro, you can see the real format.

Regards,
Ian

Joonyahny, 2024-01-10:
OK. I do not have monitorPI but will try 24/192.  And can AES play 32/44.1?  what would be the limit?

iancanada, 2024-01-10:
44.1K/32bit looks very special. The software may play it in a special
physical format. But you don't have a MonitorPi or Por so you can not see.