iancanada/DocumentDownload #11: low output on I/V transformer stage with ES9028Q2M

SolderStijn, 2024-01-22:
Hey Anybody can help me out on this...

I bought the I/V transformer stage with 1544a Lundahls for the ES9028Q2m DAC hat (using Volumio). 
Output volume is very low when feeding valve amps (input impedance 90K or so), the amp does however have a 100K volume pot as at the input (hard to drive passively?).
The VU meter hardly moves while my Phono stage makes the VU move considerably.
It seems that volume is also low when feeding the line input of a separate sound card.
Previously I used a valve output stage with the DAC (ECC83 with voltage gain of nearly 100) and the output volume was good/decent.

I have desoldered the transformers from the board and tried to use the complete winding ratio of 1:4 winding for the RCA (instead of 1:4 for XLR and 1:2 for RCA). 
Also have experimented with various combinations of primary and secondary resistors to increase voltage or to decrease output impedance. 

Does this I/V suppose to deliver the standard 2V of line level voltage? or is additional pre-amp required?
Or is there maybe an issue with a DAC setting that prevents full output volume? 
Also considering that the valve stage needed to be a mu100 valve to give decent voltage and this transformer is only a 1:4 winding ratio....

Thanks, 

Regards Stijn

SolderStijn, 2024-01-22:
Is this required with the ESS controller or can this be done differently? For example in the Volumio terminal or so?

iancanada, 2024-01-22:
My DAC always needs a controller, either a  MonitorPiPro or a ESS controller

SolderStijn, 2024-01-22:
Ahh Okay I miss interpreted the documentation. So this is definitely the issue probably? Would have saved me quite some tinkering. The manual says settings can be controlled by external ESS-controler or by Linux driver. Is the software route possible with Volumio or is the ESS-controler a better route?

SolderStijn, 2024-01-22:
Okay thanks for the information, I will buy an ess-controller, Hopefully I am pleasantly surprised by a full and rich sound.

iancanada, 2024-01-22:
BTW, can you please send me a picture showing your system so that I can
confirm what you will need.

Thanks
Ian

SolderStijn, 2024-01-23:
Btw, do you have any additional recommendations or thoughts on loading resistors (primary and secondary)? Some circuits use these, but yours only has the 510 ohm primary resistor and no R on the secondary. Will this change if I wire the 2 secondary windings in series for the RCA output? or does this change because of the input impedance of the amp in combination with the turn ratio?

SolderStijn, 2024-01-23:
Hello Ian, thanks for the quick reply, very much appreciated. I will order a MonitorPi pro and do the needed ess adjustments to 0db. Think this will likely resolve the issue, but we shall see.

iancanada, 2024-01-23:
OK

SolderStijn, 2024-01-30:
€100 later and with a Monitor PI Pro it now works! sounds great, thanks for the quick support!

iancanada, 2024-01-30:
Thank you for the update.
I'm glad you made it.
Enjoy the music and let me know at any time if there is anything I can do
to help

Regards,
Ian